moononournation/arduino-nofrendo #2: bitmap.c block corrupt ERROR

thomasgoehler, 2020-11-14:
Hi moononournation,
i have build your project which you published on "instructables".
Compiling and uploading works without problems.
When I restart the TTGO T8 v1.7 I only get a grey screen and on the serial console it shows me the following error:

addr: 0x3FFCF944, size: 61443, line 86 of C:\Users\thomas\Documents\Arduino\libraries\arduino-nofrendo-master\src\bitmap.c -- block corrupt

can you fix it?

moononournation, 2020-11-23:
the log "bitmap.c -- block corrupt" is fine. It is nofrendo normal debug message and you can show this message means nofrendo runs well.
The grey screen probably caused by the LCD connection. To narrow down the problem, you can first try to run the Arduino_GFX PDQgraphictest example for testing the display only.

closed 2022-08-28 by moononournation

lordherp, 2022-10-13:
I'm having the same problem, display is white and the "block corrupt" message. I already verified my display is working correctly.  Is there any way to debug this?

moononournation, 2022-10-13:
Can you run Arduino_GFX PDQgraphictest example correctly?

lordherp, 2022-10-13:
Yes. 
[image]

moononournation, 2022-10-13:
sorry, nes cannot display in 160x128 resolution