moononournation/arduino-nofrendo #3: Can't start game on SD Card

Sampozzo, 2021-04-15:
Hello, thank you for your great project i have big fun with it!!
My problem is that games can't start from SD card, SPIFFS is ok.

I did a little menu at boot to select the game, it show in console that i can read from SD card, but if i try to start a game the esp32 crashes and reboot.

I used your suggested pins and configuration in hw_config.h:

`#define FILESYSTEM_BEGIN SPIClass spi = SPIClass(HSPI); spi.begin(14, 2, 15, 13); SD.begin(13, spi, 8000000, FSROOT); FS filesystem = SD;`

Your can see in console that the roms can be listed:

`[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][esp32-hal-cpu.c:115] removeApbChangeCallback(): not found func=400EC25C arg=3FFBEE08
/fs/1942.nes    
/fs/Tetris.nes  
/fs/Arkanoid.nes
/fs/Galaga.nes  
_my_malloc: 256 at src\sound.c:27
_my_malloc: 980 at .pio\libdeps\az-delivery-devkit-v4\arduino-nofrendo\src\bitmap.c:48
video driver: Simple DirectMedia Layer at 256x240
_my_malloc: 15 at .pio\libdeps\az-delivery-devkit-v4\arduino-nofrendo\src\nofrendo.c:242
[W][sd_diskio.cpp:101] sdWait(): Wait Failed
[E][sd_diskio.cpp:123] sdSelectCard(): Select Failed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.      
Core 1 register dump:
PC      : 0x400ebfd8  PS      : 0x00060330  A0      : 0x800e53d0  A1      : 0x3ffb1830  
A2      : 0x00000000  A3      : 0x3ffb1899  A4      : 0x00000006  A5      : 0x00000040  
A6      : 0x00000010  A7      : 0xffffffc9  A8      : 0x00000002  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0xff000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008b9a5  LEND    : 0x4008b9b5  LCOUNT  : 0xfffffffd

ELF file SHA256: 0000000000000000

Backtrace: 0x400ebfd8:0x3ffb1830 0x400e53cd:0x3ffb1850 0x400e8f62:0x3ffb1870 0x400e919c:0x3ffb18c0 0x400e92bd:0x3ffb18f0 0x40124226:0x3ffb1930 0x40122be6:0x3ffb1950 0x40124b19:0x3ffb1990 0x40127e63:0x3ffb19c0 0x4000bdbb:0x3ffb19e0 0x4008b5c9:0x3ffb1a00 0x4008b586:0x3ffb1a20 0x40105c76:0x3ffb1a40 0x40105ce9:0x3ffb1a70 0x400d53c9:0x3ffb1a90 0x400d3339:0x3ffb1ef0 0x400d650f:0x3ffb1f10 0x400d183d:0x3ffb1f40 0x400d6497:0x3ffb1f60 0x400d1236:0x3ffb1f80 0x400ecd71:0x3ffb1fb0 0x4008d6aa:0x3ffb1fd0

Rebooting...`

I'm using an ESP32-WROOVER-B with 8mb spram and 4mb flash. 
SPIFFS works great even with big roms.

Any idea?

Thank you again.

Sampozzo, 2021-04-15:
I forget to mention that i ported your Arduino IDE example to platform.io, but it doesn't work on Arduino iDE too.

Sampozzo, 2021-04-17:
Hello, someone can help me on this?

Sampozzo, 2021-04-19:
I solved my problem removing the SD part in hw_config.h and in the main file on top::

```
#define SD_CLK 14
#define SD_MISO 2
#define SD_MOSI 15
#define SD_CS 13
SPIClass spiSD(HSPI);
```

in setup:

`//FILESYSTEM_BEGIN

  SPI.end(); 
  spiSD.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CS, spiSD))
  {
    Serial.println("Card Mount Failed");`
}
 else
  {
    Serial.println("Card Mount Successful");
  }
  File root = SD.open("/");
`

I don't know why this works for me. I hope this helps someone with my same problem.

linkov, 2022-05-24:
@[Sampozzo](https://github.com/Sampozzo) Can you please share your project, I added M5Stack_TreeView to choose rom and now I get

```
NoFrendo start!

_my_malloc: 256 at src/sound.c:26
_my_malloc: 980 at src/noufrendo/bitmap.c:48
video driver: Simple DirectMedia Layer at 256x240
_my_malloc: 12 at src/noufrendo/nofrendo.c:242
_my_malloc: 12 at src/noufrendo/nofrendo.c:152
system type unknown, playing nofrendo NES intro.
_my_malloc: 12 at src/noufrendo/nofrendo.c:152
_my_malloc: 808 at src/noufrendo/nes/nes.c:522
_my_malloc: 92 at src/noufrendo/nes/nes.c:539
_my_malloc: 2048 at src/noufrendo/nes/nes.c:546
_my_malloc: 296 at src/noufrendo/sndhrdw/nes_apu.c:990
_my_malloc: 7584 at src/noufrendo/nes/nes_ppu.c:127
_my_malloc: 1068 at src/noufrendo/nes/nes_rom.c:426
GUI: 3DBlock.nes not found, will use default ROM
_my_malloc: 0 at src/noufrendo/nes/nes_rom.c:131
_my_malloc: 16384 at src/noufrendo/intro.c:4451
_my_malloc: 8192 at src/noufrendo/intro.c:4471
GUI: ROM loaded: Nofrendo intro [0] 16k/8k V
_my_malloc: 8 at src/noufrendo/nes/nes_mmc.c:241
created memory mapper: None
setting up mapper 0
reset memory mapper
GUI: NES powered on
_my_malloc: 61440 at src/noufrendo/bitmap.c:86
ASSERT: line 267 of src/noufrendo/memguard.c, malloc: out of memory at line 86 of src/noufrendo/bitmap.c.  block size: 61440

memory leak - 11 unfreed blocks

addr: 0x3FFB28D8, size: 256, line 26 of src/sound.c
addr: 0x3FFAF110, size: 808, line 522 of src/noufrendo/nes/nes.c
addr: 0x3FFB6300, size: 92, line 539 of src/noufrendo/nes/nes.c
addr: 0x3FFAF488, size: 2048, line 546 of src/noufrendo/nes/nes.c
addr: 0x3FFAFCD8, size: 296, line 990 of src/noufrendo/sndhrdw/nes_apu.c
addr: 0x3FFD0634, size: 7584, line 127 of src/noufrendo/nes/nes_ppu.c
addr: 0x3FFB7408, size: 1068, line 426 of src/noufrendo/nes/nes_rom.c
addr: 0x3FFAFE50, size: 0, line 131 of src/noufrendo/nes/nes_rom.c
addr: 0x3FFD2424, size: 16384, line 4451 of src/noufrendo/intro.c
addr: 0x3FFD6474, size: 8192, line 4471 of src/noufrendo/intro.c
addr: 0x3FFAFEA0, size: 8, line 241 of src/noufrendo/nes/nes_mmc.c

abort() was called at PC 0x4010084e on core 1

Backtrace:0x40083a29:0x3ffb24200x40089f71:0x3ffb2440 0x4008f2cd:0x3ffb2460 0x4010084e:0x3ffb24e0 0x4010f067:0x3ffb2500 0x400e2cca:0x3ffb2520 0x400e51f6:0x3ffb2540 0x400d92aa:0x3ffb2680 0x400e9916:0x3ffb26a0 0x400e8483:0x3ffb26c0 0x400e9b5e:0x3ffb26f0 0x400e83bd:0x3ffb2710 0x400d91c2:0x3ffb2730 0x4012053b:0x3ffb2780 0x400d3b49:0x3ffb27a0 0x400d32ab:0x3ffb27d0 0x400d8d41:0x3ffb2800 0x400f177d:0x3ffb2820 
  #0  0x40083a29:0x3ffb24200 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

```

Sampozzo, 2022-05-26:
Hi @linkov i shared a little project based on arduino-nofredo ready for paltform.io. It loads a game from spiffs, but it is simple to use SD too, i will share and example as soon as i can.

You can find it here: https://github.com/ZioTester/NESP32
Sorry but blog post is in italian language.

linkov, 2022-05-28:
@Sampozzo thanks for sharing, your devices look super cool :D I could not figure out how to make it work on M5Stack Grey, its probably due to the fact that it requires special settings due to SD and TFT using same resources. M5Stack-nesemu is is the only repo that works but as soon as I add Arduino the display stops working

linkov, 2022-06-01:
@Sampozzo hey, figured it out, thanks for your support! :D . You can check the repo here: https://github.com/FvnctionHQ/m5stack-nes

closed 2022-08-28 by moononournation