moononournation/arduino-nofrendo #6: M5Stack Core Esp32 - Doesn't work

Dario-Ciceri, 2022-08-27:
As described, M5Stack Core Esp32 with Esp32 Core v2.0.4 doesn't work. It keeps resetting!!!

moononournation, 2022-08-28:
At the time I am writing this, the Arduino-esp32 still at version 1.0.x. Please try the older Arduino-esp32 v1.0.6 in Arduino IDE working first.

Dario-Ciceri, 2022-08-28:
> At the time I am writing this, the Arduino-esp32 still at version 1.0.x. Please try the older Arduino-esp32 v1.0.6 in Arduino IDE working first.

Ok I try now, btw I had to rename the display initializer in Arduino_ILI9342 *gfx = new Arduino_ILI9342(bus, 33 /* RST */, 2 /* rotation */); because you updated the gfx library but not the NES code

Dario-Ciceri, 2023-07-31:
Hi, back again on this issue and I found the error code `addr: 0x3FFE8C34, size: 61440, line 86 of C:\Users\cicer\Documents\Arduino\libraries\arduino-nofrendo-master\src\bitmap.c -- block corrupt`
which is: `addr = NOFRENDO_MALLOC(((pitch * height) + 3) & 0xFFFFFFF8); /* add max 32-bit aligned adjustment */`

moononournation, 2023-08-01:
that means cabnot allocat memory, you should use PSRAM

Dario-Ciceri, 2023-08-01:
> that means cabnot allocat memory, you should use PSRAM

On arduino IDE I can't find psram enable for M5stack, also I tried with esp-idf but there's no cmakelist

moononournation, 2023-08-01:
or try another smaller ROM

Dario-Ciceri, 2023-08-01:
> or try another smaller ROM

What if I use SD card?

moononournation, 2023-08-01:
I mean smaller game, the game ROM need load into RAM before start.

Dario-Ciceri, 2023-08-01:
> I mean smaller game, the game ROM need load into RAM before start.

Yes this is fine, I tried without a game ROM and it gives the same problem, I think it's a game ROM problem

Dario-Ciceri, 2023-08-01:
What is it trying to load after showing nofrendo logo?
I still get `addr: 0x3FFE8C34, size: 61440, line 86 of C:\Users\cicer\Documents\Arduino\libraries\arduino-nofrendo-master\src\bitmap.c -- block corrupt`

Dario-Ciceri, 2023-08-01:
I tried loading tetris.nes, it detects it but after nofrendo intro I get: GUI: /fstetris.nes not found, will use default ROM and block corruption

moononournation, 2023-08-01:
seems file path missing a '/' in the middle.

Dario-Ciceri, 2023-08-01:
> seems file path missing a '/' in the middle.

should I rename tetris.nes to /tetris.nes?

Dario-Ciceri, 2023-08-01:
> seems file path missing a '/' in the middle.

So what should I change?

Dario-Ciceri, 2023-08-01:
nothing seems to works, I can't find where's the path it's trying use

and3rson, 2024-02-01:
> seems file path missing a '/' in the middle.

I had the same issue, had to add "/" in `main.cpp`:

```cpp
                    sprintf(fullFilename, "%s/%s", FSROOT, filename);
```