moononournation/arduino-nofrendo #8: what print when the nes game work sucsessful?

justdomyself, 2023-04-11:
Like this ?

my_malloc: 808 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes.c:522
_my_malloc: 92 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes.c:539
_my_malloc: 2048 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes.c:546
_my_malloc: 296 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\sndhrdw\nes_apu.c:990
_my_malloc: 7584 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes_ppu.c:127
_my_malloc: 1068 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes_rom.c:426
_my_malloc: 8192 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes_rom.c:131
GUI: ROM loaded: Chase.nes [0] 16k/8k H
_my_malloc: 8 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\nes\nes_mmc.c:241
created memory mapper: None
setting up mapper 0
reset memory mapper
GUI: NES powered on
_my_malloc: 61440 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\bitmap.c:86
_my_malloc: 980 at C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\arduino-nofrendo-master\src\bitmap.c:48

davidstrap, 2023-04-12:
I have the same question cause i don't see nothing but a black screen, also the input (from joystick) is read just one time, so i don't know if it's working or not:

_my_malloc: 980 at ../nes/bitmap.c:48
video driver: ST7796S at 256x240
_my_malloc: 10 at ../nes/nofrendo.c:246
_my_malloc: 10 at ../nes/nofrendo.c:155
_my_malloc: 808 at ../nes/nes/nes.c:521
_my_malloc: 92 at ../nes/nes/nes.c:538
_my_malloc: 2048 at ../nes/nes/nes.c:545
_my_malloc: 296 at ../nes/sndhrdw/nes_apu.c:990
_my_malloc: 7584 at ../nes/nes/nes_ppu.c:127
_my_malloc: 1064 at ../nes/nes/nes_rom.c:638
_my_malloc: 8192 at ../nes/nes/nes_rom.c:132
GUI: ROM loaded: CHASE.NES [0] 16k/8k H
_my_malloc: 8 at ../nes/nes/nes_mmc.c:241
created memory mapper: None
setting up mapper 0
reset memory mapper
GUI: NES powered on
_my_malloc: 61440 at ../nes/bitmap.c:86
_my_malloc: 980 at ../nes/bitmap.c:48
Timer install, configTICK_RATE_HZ=100, freq=60
Input: ffffffff

So any help would be appreaciated

davidstrap, 2023-04-13:
Finally i get something in the terminal that suggest is not working
GUI: error: could not open SNSS file
As far as i can tell, it could not read the file CONTRA.SS0, so any help would be appreciated.

davidstrap, 2023-04-13:
Now i'm getting kinda troubles with save0, seems to be corrupted :-/ 
GUI: State 0 saved
GUI: error: not an SNSS file

justdomyself, 2023-04-28:
> 
can you see backlight ?

davidstrap, 2023-04-28:
> 

Hi there, 
Yep, everything works as expected, 
Ty :-D

closed 2024-05-12 by moononournation