moononournation/arduino-nofrendo #14: Memory leaks - DOIT ESP32 DEVKIT V1

Jailin, 2023-12-17:
Getting the following error:

memory leak - 11 unfreed blocks

addr: 0x3FFBBEF8, size: 256, line 26 of /home/jailin/.var/app/cc.arduino.IDE2/cache/.arduinoIDE-unsaved20231118-51-zk56s0.30hm/esp32-nofrendo/sound.c
addr: 0x3FFD9128, size: 808, line 522 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c
addr: 0x3FFD94A0, size: 92, line 539 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c
addr: 0x3FFD954C, size: 2048, line 546 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c
addr: 0x3FFD9D9C, size: 296, line 990 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/sndhrdw/nes_apu.c
addr: 0x3FFD9F14, size: 7584, line 127 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_ppu.c
addr: 0x3FFDBD04, size: 1068, line 426 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_rom.c
addr: 0x3FFDC180, size: 0, line 131 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_rom.c
addr: 0x3FFF46F4, size: 16384, line 4451 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/intro.c
addr: 0x3FFDC1D0, size: 8192, line 4471 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/intro.c
addr: 0x3FFDE220, size: 8, line 241 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_mmc.c

abort() was called at PC 0x400fc50a on core 1

Backtrace: 0x40083855:0x3ffce650 0x4008bfa1:0x3ffce670 0x400919bd:0x3ffce690 0x400fc50a:0x3ffce710 0x40122fc3:0x3ffce730 0x400e2572:0x3ffce750 0x400e4a9e:0x3ffce770 0x400d8b52:0x3ffce8b0 0x400e91be:0x3ffce8d0 0x400e7d2b:0x3ffce8f0 0x400d30a2:0x3ffce920 0x400e7c65:0x3ffce940 0x400d2e3a:0x3ffce960 0x400ebe9e:0x3ffceb10

Samighz, 2024-02-24:
same problem with me

roundspring2003, 2024-03-08:
same problem with me

justdomyself, 2024-05-16:
> Getting the following error:
> 
> _my_malloc: 61440 at /home/jailin/Arduino/libraries/arduino-nofrendo/src/bitmap.c:86 ASSERT: line 266 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/memguard.c, malloc: out of memory at line 86 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/bitmap.c. block size: 61440
> 
> memory leak - 11 unfreed blocks
> 
> addr: 0x3FFBBEF8, size: 256, line 26 of /home/jailin/.var/app/cc.arduino.IDE2/cache/.arduinoIDE-unsaved20231118-51-zk56s0.30hm/esp32-nofrendo/sound.c addr: 0x3FFD9128, size: 808, line 522 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c addr: 0x3FFD94A0, size: 92, line 539 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c addr: 0x3FFD954C, size: 2048, line 546 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes.c addr: 0x3FFD9D9C, size: 296, line 990 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/sndhrdw/nes_apu.c addr: 0x3FFD9F14, size: 7584, line 127 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_ppu.c addr: 0x3FFDBD04, size: 1068, line 426 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_rom.c addr: 0x3FFDC180, size: 0, line 131 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_rom.c addr: 0x3FFF46F4, size: 16384, line 4451 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/intro.c addr: 0x3FFDC1D0, size: 8192, line 4471 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/intro.c addr: 0x3FFDE220, size: 8, line 241 of /home/jailin/Arduino/libraries/arduino-nofrendo/src/nes/nes_mmc.c
> 
> abort() was called at PC 0x400fc50a on core 1
> 
> Backtrace: 0x40083855:0x3ffce650 0x4008bfa1:0x3ffce670 0x400919bd:0x3ffce690 0x400fc50a:0x3ffce710 0x40122fc3:0x3ffce730 0x400e2572:0x3ffce750 0x400e4a9e:0x3ffce770 0x400d8b52:0x3ffce8b0 0x400e91be:0x3ffce8d0 0x400e7d2b:0x3ffce8f0 0x400d30a2:0x3ffce920 0x400e7c65:0x3ffce940 0x400d2e3a:0x3ffce960 0x400ebe9e:0x3ffceb10

------------>
out of memory at line 86 of ...

the reseson is out if memory.  the ram of the chip is not enough

Suggest use esp32s3 n16r8  ,and enable the psram , then you can run large nes rom games.